RMerl/asuswrt-merlin.ng #746: cru not work as expected when expect it to run once every two days.

zw963, 2021-04-20:
**Describe the bug**

Add a task,  i want it to run once every two days, so i add a new task like this.

```
cru a update_geosites "* * */2 * *" "/opt/etc/update_geosite.sh"
```

What i expected is, "/opt/etc/update_geosite.sh" will be run once every two days,
but after i add it, it will be run very frequently which cause network disconnect.
(because update_geosite.sh will restart network)

following is content of update_geosite.sh

https://raw.githubusercontent.com/zw963/asuswrt-merlin-v2ray/master/router/opt/etc/update_geosite.sh

**Model**

RT-AC5300

**Firmware version**

384.19

**Expected behavior**

It should not run so frequently.

paldier, 2021-04-20:
>  (because update_geosite.sh will restart network)

Why?  You do not need to restart network.

zw963, 2021-04-20:
> The command to restart dnsmasq is `service restart_dnsmasq`

Sorry, the key issue is not restart network, the issue is `/opt/etc/update_geosite.sh` was restart too frequently.

In fact, what i means for restart network is this. `/opt/etc/init.d/S22v2ray restart`, it is a proxy service.

paldier, 2021-04-20:
You do NOT NEED to restart network, OK? V2ray will not do that, it's you. So, the firmware has no bugs.
If you want to learn how to restart plugins, you can see https://github.com/MerlinRdev/softcenterarm/blob/master/koolproxy/koolproxy/koolproxy/kp_config.sh

zw963, 2021-04-20:
> You do NOT NEED to restart network, OK? V2ray will not do that, it's you. So, the firmware has no bugs.
> If you want to learn how to restart plugins, you can see https://github.com/MerlinRdev/softcenterarm/blob/master/koolproxy/koolproxy/koolproxy/kp_config.sh

Sorry,  maybe not explain clearly,  i am not restart network, i just restart v2ray to load newest `geosite.dat`.
but when restart v2ray, my network is disconnect because proxy is temorary unvailable.

But anyway, above things not relative to this issue, the issue is, why this script has been reboot frequently instead of once two day?

kwloafman, 2021-04-20:
See here
<https://www.cyberciti.biz/faq/how-to-add-cron-job-on-asuswrt-merlin-wifi-router/>.
Your command should be:

# cru a update_geosites "* * */2 * * /opt/etc/update_geosite.sh"

all as one string, not two.

...Ken

paldier, 2021-04-20:
````
Cron Utility
add:    cru a <unique id> <"min hour day month week command">
delete: cru d <unique id>
list:   cru l
````

zw963, 2021-04-20:
Thank  you, following is my cron tasks.

[image]

the first one is run with  `cru a update_geosites "* * */2 * *" "/opt/etc/update_geosite.sh"`
the second one is run with `cru a update_geosites "* * */2 * * /opt/etc/update_geosite.sh"`

Any different ? 

in fact, i think another two service(run-services, run-iptables) is works as expected.

paldier, 2021-04-20:
Are you kidding me? 
````
*/1 * * * * /jffs/scripts/services-start
````

zw963, 2021-04-20:
Hi, i tested with new syntax,  `cru a update_geosites1 "* * */2 * * /opt/etc/update_geosite.sh"`, it not work too.

I am very sure `/opt/etc/update_geosite.sh` is invoked very frequently, so, that is a bug?

 

> Are you kidding me?
> 
> ```
> */1 * * * * /jffs/scripts/services-start
> ```

No, i hope `services-start` can be run frequently to ensure sometimes somethings broken v2ray, as you can see, it run very quickly if all program was working fine.

[image]

RMerl, 2021-04-20:
> No, i hope services-start can be run frequently to ensure sometimes somethings broken v2ray, as you can see, it run very quickly if all program was working fine.

Do NOT do that.  That script is only intended to be run ONCE, at **boot time**.

Your cron entry does not work because busybox's crontab does not allow using the "/" operator on the day, it's only support on minutes and hours.  See this reference:

https://openwrt.org/docs/guide-user/base-system/cron

This is working as intended.  If you want extended crontab support, you will have to contact the busybox devs about this.

closed 2021-04-20 by RMerl