RMerl/asuswrt-merlin.ng #748: Firmware Upgrade: AiMesh Node not listed

Ruslan-Urban, 2021-05-10:
**Description**
I was able to upgrade the main router to the latest firmware version 386.2_4 as of May 10, 2021. I have a main router and one mesh node, but firmware upgrade option does not show up for the node as described at [How to update firmware of my AiMesh node?](https://www.asus.com/us/support/FAQ/1035199/).

**Model/Firmware version**
Main router: RT-AC66U B1/Firmware Version 386.2_4
AiMesh node: RT-AC68U/Firmware Version 386.1_2

**To Reproduce**
Navigate to Administration > Firmware Upgrade.
Alternatively, click on the link below the yellow exclamation mark in the top-right corner.

**Expected behavior**
Firmware upgrade options should be available for all AiMesh nodes. It appears that the only option to upgrade the firmware is to remove the node and re-add it back after the firmware is upgraded.

**Actual results**
Firmware upgrade option is not listed

**Screenshots**
[image]
[image]

genius3000, 2021-05-10:
The version on your node implies it's running Asuswrt-Merlin which has a limitation compared to stock when it comes to upgrading. The second paragraph of the Wiki page describes this. How to manually upgrade isn't explained in depth there but likely is on the support forums. fwiw, I just run the recommendation of stock firmware of the nodes.

RMerl, 2021-05-10:
Asuswrt-Merlin does not support live updates.  You have to click on the "Upload" link on the Firmware Upgrade page, and manually update updates.

closed 2021-05-10 by RMerl

Ruslan-Urban, 2021-05-10:
There is no option to manually update the firmware for the node as the [Wiki page](https://github.com/RMerl/asuswrt-merlin.ng/wiki/AiMesh) implies. There is no "Upload" link for the node. You can see it in the screenshot. 
When you navigate to the the node's IP address, you get automatically redirected to the main router URL. Thus, there is no way to update the firmware, except by removing the node from the mesh.