RMerl/asuswrt-merlin.ng #749: Development questions

galak, 2021-05-12:
Had a few questions related to development:
* Are fixes associated with building on different distros something you take upstream (for example was building RT-AX86U firmware on Fedora 33 and ran into a few packages that fail to build because of aclocal-1.16.
* How do sync's with new source drops from Asus work?

RMerl, 2021-05-12:
No, because of the complexity of the build system and also to make future GPL merges as easy as possible, I try to avoid making changes to the build system (which includes Makefiles, configure, etc...).

New GPLs are merged by hand, by doing a three-way diff between the previous GPL, the new GPL and my own code.  It must be done directory by directory (as some specific directories cannot be merged), and I must manually resolve conflicts (which generally means manually merging the conflicting portions line by line).  And at the end, I use a script that copies the prebuilt binary blobs to the appropriate locations (has to be done for each individual models).

Basically...  A lot of manual work involved with any new GPL release, not something that can easily be documented and explained.

galak, 2021-05-12:
@RMerl Here's an example of a change I made to get a sense if its interesting to send a PR or not:

```
diff --git a/release/src/router/Makefile b/release/src/router/Makefile
index a1fc6b2cef..afbfa85d6e 100644
--- a/release/src/router/Makefile
+++ b/release/src/router/Makefile
@@ -7164,10 +7164,11 @@ GeoIP-1.6.2-clean:
        -@[ ! -f GeoIP-1.6.2/Makefile ] || $(MAKE) -C GeoIP-1.6.2 distclean
        @rm -f GeoIP-1.6.2/stamp-h1
 
-wget/Makefile.in: wget/Makefile.am
+wget/stamp-h1:
        cd wget && autoreconf -fi
+       touch $@
 
-wget/Makefile: wget/Makefile.in
+wget/Makefile: wget/stamp-h1
        cd wget && $(CONFIGURE) \
                --with-ssl=openssl --prefix=/usr --sysconfdir=/etc \
                --disable-opie --disable-ntlm --disable-debug --disable-nls --disable-rpath \
@@ -7186,6 +7187,7 @@ wget: openssl zlib wget/Makefile
 wget-clean:
        [ ! -f wget/Makefile ] || $(MAKE) -C wget distclean
        @rm -f wget/Makefile
+       @rm -f wget/stamp-h1
```

RMerl, 2021-05-12:
The problem with these is they will make it harder to do a diff/patch without having to manually resolve conflicts.  I try to limit changes to only those that are actually important - support for different build environment is unfortunately not important enough for me to consider merging a PR, sorry.

snaunton, 2021-05-12:
@galak build inside an Ubuntu docker image, then it doesn't matter what platform you are using.

closed 2021-07-02 by RMerl