RMerl/asuswrt-merlin.ng #751: NTP and VPN deadlock ?

IAmTheNail, 2021-05-22:
**Describe the bug**
The VPN client couldn't connect with my VPN provider because its certificate was not yet valid. At the same time NTP was failing to set the clock to the real time. I assume both the services were holding each other in a deadlock - The VPN client needing the clock to be set right, while the NTP service was failing to connect to the ntp provider.

This happened after a scheduled reboot. Maybe that's important to mention.

The workaround is to shut down the VPN client long enough for the NTP service to set the proper time. With this confirming the VPN client somehow blocked the NTP service. 

The VPN client should not start before the NTP service successful clock sync.

**Model**
RT-AC88U

**Firmware version**
386.2_4 (latest at the time of writing)

**Client information**

**To Reproduce**
I did manual reboot in the past without errors. Maybe those were lucky breaks ? As long as the VPN provider's certificate is valid according to the clock, whatever time that was.

**Expected behavior**
The clock to set right & VPN client connected to the VPN provider.

**Actual results**
The clock set to a time in the past & the VPN client failing to connect to the VPN provider.

**Will this error be reproduced on the official firmware?**
I don't think so because I had this configuration in the past already.

**Screenshots**

**Additional context**
Two reoccurring lines in the log say it all:

```
May  5 13:24:45 reboot_scheduler: [timecheck] NTP sync error
May  5 13:25:04 ovpn-client3[19997]: VERIFY ERROR: depth=0, error=certificate is not yet valid: ....
```

The timestamps are 16 days off, as a consequence the VPN client refuses the provider's certificate. And I guess the NTP service is unable to connect to the pool.ntp.org because the VPN client is busy.

sparkym3, 2021-05-22:
I have been dealing with similar on my RT-AC86U for a while now and usually just disconnect WAN and reconnect to trigger the ntp update. I am using VPN, DoT, and local ntp server which I assume all plays into what is going on here.

cyborgxxl, 2021-06-02:
> **Describe the bug**
> The VPN client couldn't connect with my VPN provider because its certificate was not yet valid. At the same time NTP was failing to set the clock to the real time. I assume both [...]

I have the same issue....