RMerl/asuswrt-merlin.ng #752: [bug] smb signing caused high cpu usage

rhjdvsgsgks, 2021-05-22:
**Describe the bug**
high cpu usage when copying file from inbuilt samba server

**Model**
ac86u

**Firmware version**
386.2_4

**Client information**
normal smb2 client

**To Reproduce**
copying file from inbuilt samba server

**Expected behavior**
acceptable cpu usage

**Actual results**
excessive high cpu usage (one core 100%)

**Will this error be reproduced on the official firmware?**

**Screenshots**

**Additional context**
this is a very classical bug, exist a lot of time, but no provided actual solution ("you should buy a nas" is not a "answer"). and now i found it, high cpu usage is caused by smb signing, after disabling it by add
```
client ipc signing = disabled
client signing = disabled
server signing = disabled
```
to smb.conf (before path config part) , cpu usage significantly decreased (100% -> about 10%). so is it possible to add these lines to default smb config?

JackMerlin, 2021-05-23:
Have you checked whether the transmission speed is affected after modifying ``smb.conf``?

Because if you only use 10% CPU, it means that there is 90% room for improvement in transmission speed. I think @Adamm00 will be very interested in this, because he found that modifying the socket to improve the transmission speed, and I made a [script](https://github.com/JackMerlin/USBAccelerator).

RMerl, 2021-05-23:
What exact client are you using?  With Windows 10, I am able to reach my full Gigabit rate of 110 MB/s without the need of any config change with my RT-AX88U.  The Samba process only hits around 15% usage, 22% usage for the Broadcom inbound process, and 37% system usage (the USB subsystem).

RMerl, 2021-05-23:
With Windows, you can check whether you are using a signed connection with this Powershell cmdlet (with default settings, my Windows 10 PC doesn't):

```
PS C:\WINDOWS\system32> Get-SmbConnection stargate88ax | Select-Object -Property *

SmbInstance           : Default
ContinuouslyAvailable : False
Credential            : CAMELOT\admin
Dialect               : 2.0.2
Encrypted             : False
NumOpens              : 1
Redirected            : False
ServerName            : stargate88ax
ShareName             : stored
Signed                : False
UserName              : CAMELOT\Merlin
PSComputerName        :
CimClass              : ROOT/Microsoft/Windows/SMB:MSFT_SmbConnection
CimInstanceProperties : {ContinuouslyAvailable, Credential, Dialect, Encrypted...}
CimSystemProperties   : Microsoft.Management.Infrastructure.CimSystemProperties
```

rhjdvsgsgks, 2021-05-23:
> What exact client are you using? With Windows 10, I am able to reach my full Gigabit rate of 110 MB/s without the need of any config change with my RT-AX88U. The Samba process only hits around 15% usage, 22% usage for the Broadcom inbound process, and 37% system usage (the USB subsystem).

I'm a Android user, using a file manager which used [smbj](https://github.com/hierynomus/smbj) as smb client

rhjdvsgsgks, 2021-05-23:
> Have you checked whether the transmission speed is affected after modifying `smb.conf`?
> 
> Because if you only use 10% CPU, it means that there is 90% room for improvement in transmission speed. I think @Adamm00 will be very interested in this, because he found that modifying the socket to improve the transmission speed, and I made a [script](https://github.com/JackMerlin/USBAccelerator).
> 
> I also recommend that you post your findings on the [SNB forum](https://www.snbforums.com/forums/asuswrt-merlin.42/), because this will be very helpful to many people, even if it may not be included in the firmware in the short term.

have a little bit improve (2m to 3m), but still far from ftp transmission speed (10m, close to "Receive link speed"(due a bug, max download speed of my device was limited to this)),  and cpu cost also higher than ftp(proximately zero) even if adding this config

RMerl, 2021-05-24:
Check if your client has an option to disable signing, as it seems to be enforcing it, which differs from a standard Windows client.

Force disabling it at the router level would be a last resort thing, as it means any client that make signing mandatory would no longer work at all if the router completely disabled signing support.

rhjdvsgsgks, 2021-05-25:
> Check if your client has an option to disable signing, as it seems to be enforcing it, which differs from a standard Windows client.

unfortunately, most of Android smb client didn't provide this option, that's why i disable it on server side.

> Force disabling it at the router level would be a last resort thing, as it means any client that make signing mandatory would no longer work at all if the router completely disabled signing support.

so, should I close this issue?

RMerl, 2021-07-02:
Since changing default behaviour risks completely breaking access for clients that make it mandatory, I will leave it as it is.  Users who wish to disable it can do so through a custom smb configuration.

closed 2021-07-02 by RMerl