RMerl/asuswrt-merlin.ng #754: Can have at most 1 VPN client connect to the firmware's OpenVPN server

changyuheng, 2021-05-25:
If multiple clients connect to the firmware's OpenVPN server, only the latest connected client will work, the others will get stalled.

haohetao, 2021-07-12:
I have same problem,after each reboot, only first openvpn client can connect,all other client authentication failed.

OliverPA77, 2021-07-17:
Had the same issue and tried to generate further client certificates following the [guide](https://github.com/RMerl/asuswrt-merlin.ng/wiki/Generating-OpenVPN-keys-using-Easy-RSA) in the wiki but it doesn't work (lots of errors coming from the openssl.cnf file). I assume we'd need to generate additional client certificates and then edit the downloaded ovpn files.

As a workaround I made two changes - after that having >1 client connecting worked fine and survived a reboot.
VPN Server -> VPN Details -> Advanced -> Username / Password Auth. Only = Yes :(
I assume this is actually enough but for the record I also assigned static IPs by following this [guide](https://github.com/RMerl/asuswrt-merlin.ng/wiki/Static-ip-for-OpenVPN-clients).

RMerl, 2021-07-18:
Works for me.  Must be a configuration issue, look at your system log for more info.  For instance if you use client certificates, then each client must have a certificate with a unique common-name.

RMerl, 2021-07-18:
Also based on your screnshots, this isn't my firmware, this is Asus' stock firmware.  If you have an issue with the stock firmware, then contact Asus about it.

closed 2021-07-18 by RMerl

haohetao, 2021-07-19:
> Works for me. Must be a configuration issue, look at your system log for more info. For instance if you use client certificates, then each client must have a certificate with a unique common-name.

Are you saying that each user should use a separate client certificate?

haohetao, 2021-07-19:
@RMerl 
I‘m config username-as-comman-name.

OliverPA77, 2021-07-19:
Yes you need to generate a certificate for each client using this [guide](https://github.com/RMerl/asuswrt-merlin.ng/wiki/Generating-OpenVPN-keys-using-Easy-RSA), then you export the generic client config using VPN Server -> Export OpenVPN configuration file and edit it for each client (you will find "insert client certificate/key" lines in the downloaded config).
Afterwards you can also set Username/Password Authentication to No.

@RMerl The issue with the guide (or the easy rsa script) is that after the step "cd /mnt/sda1/easy-rsa" you also need to cp openssl-1.0.0.cnf openssl.cnf otherwise the script complains about missing openssl.cnf file.
Once doing this extra step everything worked as expected and as documented.

haohetao, 2021-07-26:
@RMerl 
The above problems still occur as long as the user password authentication is enabled

haohetao, 2021-07-26:
[image]