RMerl/asuswrt-merlin.ng #758: Guest Network borks Fios ONT

thejohnha, 2021-06-26:
This took me quite some time to figure out, but when I have a device connect to a guest [wireless] network, it borks the Verizon Fios ONT to the point where I need to reboot the ONT. The guest network I have has Intranet access disabled. (I haven't tested it with Intranet access enabled.)

I'm using an ASUS RT-AC88U on firmware 386.2_6

**To Reproduce**
Enable wireless guest network (only tested with Intranet access disabled). I tried with an android tablet, as well as an iPhone. As soon as the device connects, the Fios ONT becomes unusable. I can recover with the Fios app on my phone (connected to the cell network) and reboot the ONT. Or I can power cycle the ONT.

**Will this error be reproduced on the official firmware?**
I have only used Merlin firmware...

Doing various things while SSH'd into the router didn't help. I tried:
```
killall dnsmasq
rm /tmp/var/lib/misc/dnsmasq.leases
service restart_dnsmasq
service restart_firewall
service restart_upnp
service restart_httpd
```

Many thanks - this was a strange one... Googling for the issue didn't yield me anything useful 🙏

dave14305, 2021-06-26:
See https://www.snbforums.com/threads/asuswrt-merlin-386-2_6-is-now-available.72962/page-10#post-696360

thejohnha, 2021-06-28:
Thank you! I wonder if this will be "fixed" in a future release, or if the additional entries to the firewall need to be a permanent adjustment.