RMerl/asuswrt-merlin.ng #761: RT-AX86U - unmount user script not running during reboot.

ColinTaylorUK, 2021-06-30:
After pressing the "Reboot" button in the GUI the unmount user script is not running during shutdown. The services-stop script _is_ running however.

This sounds very similar to an issue that was identified in 2018: http://www.snbforums.com/threads/unmount-script-not-running-on-ac86u.50325/post-450232

**Model**
RT-AX86U. Two other people in the forums have reported the same issue specifically with this model.

http://www.snbforums.com/threads/usb-dcl-disk-check-log-constantly-recovering-journal.73336/post-697294

**Firmware version**
386.2_6

**To Reproduce**
I put statements in the services-stop and unmount that created timestamped files in /jffs. They also wrote meesages to the syslog.

**Expected behavior**
Both user scripts should run and create the messages in syslog and files in /jffs.

**Actual results**
services-stop ran as expected. There was no indication that unmount ran at all.

[reboot.log](https://github.com/RMerl/asuswrt-merlin.ng/files/6737409/reboot.log)
[jffs.txt](https://github.com/RMerl/asuswrt-merlin.ng/files/6737445/jffs.txt)

RMerl, 2021-06-30:
Works for me on an RT-AC66U_B1.

```
Jun 29 22:58:16 custom_script: Running /jffs/scripts/unmount (args: /tmp/mnt/Entware)
Jun 29 22:58:17 syslog: USB partition unmounted from /tmp/mnt/Entware
```

ColinTaylorUK, 2021-06-30:
Yes. As I said, this problem appears to be specific to the RT-AX86U.

RMerl, 2021-06-30:
Your shutdown log is different from my own, but my first guess would be a crash occuring during the shutdown.  Since it's specific to the RT-AX86U there's a good chance the crash might occur in one of the BCM SDK components.

ColinTaylorUK, 2021-06-30:
While the BCM components on this model seem prone to crash for the slightest reason it is usually indicated in the startup syslog and crashdump file. In this case there are no indications that a crash has occurred, although that doesn't mean it hasn't.

RMerl, 2021-06-30:
My own RT-AX86U either crashed or lost a good chunk of the system log.

```
Jun 30 12:11:04 kernel: UBIFS (ubi2:0): background thread "ubifs_bgt2_0" stops
Jun 30 12:11:04 kernel: ubi2: detaching mtd10
Jun 30 12:11:04 kernel: ubi2: mtd10 is detached
Jun 30 12:11:04 get_ext_phy_id: 0/1/0/0
Jun 30 12:11:05 FTP_Server: daemon is stopped
Jun 30 12:11:05 Samba_Server: smb daemon is stopped
Jun 30 12:11:05 Timemachine: daemon is stopped
Jun 30 12:11:05 FTP_Server: daemon is stopped
Jun 30 12:11:05 Samba_Server: smb daemon is stopped
Jun 30 12:11:07 Timemachine: daemon is stopped
Jun 30 12:11:07 avahi-daemon[3040]: WARNING: No NSS support for mDNS detected, consider installing nss-mdns!
Jun 30 12:11:07 WEBDAV_Server: daemon is stopped
May  5 01:05:10 kernel: klogd started: BusyBox v1.25.1 (2021-06-23 00:43:53 EDT)
```
If it's indeed a crash in an SDK component, then it's outside of my control and will have to be re-tested with the next GPL merge, which is supposed to contain an updated SDK for that platform.

ColinTaylorUK, 2021-06-30:
Is there any kind of extended debugging I can turn on to get more visibility into the shutdown flow?

As I said earlier it sounds very much like the problem from 2018, but I don't know what exactly John identified as the cause in that case so it might be completely different.

RMerl, 2021-06-30:
The only way would be to disassemble the router, and connect to its serial port to get the debug output during the reboot.

I haven't disassembled my RT-AX86U yet, will have to hope it's as easy to disassemble as the RT-AC86U (there are a few models I have given up on figuring out how to disassemble them, can't remember if the RT-AX86U was one of them).

Otherwise, would have to inject ton of debugging output to the code and hopefully get some of that output before the crash.

No ETA on next GPL, only that the original target date of end of June for Asus providing them has been delayed until the middle of July.

RMerl, 2021-07-01:
The issue is actually caused by John's fix.  Reverting that fix now allows disks to be properly enumerated and unmounted at shutdown:

```Jul  1 00:45:18 kernel: ubi2: detaching mtd10
Jul  1 00:45:18 kernel: ubi2: mtd10 is detached
Jul  1 00:45:18 get_ext_phy_id: 0/1/0/0
Jul  1 00:45:18 STOP: bc
Jul  1 00:45:18 FTP_Server: daemon is stopped
Jul  1 00:45:18 Samba_Server: smb daemon is stopped
Jul  1 00:45:18 Timemachine: daemon is stopped
Jul  1 00:45:19 custom_script: Running /jffs/scripts/unmount (args: /tmp/mnt/sda1)
**Jul  1 00:45:20 init: USB partition unmounted from /tmp/mnt/sda1**
Jul  1 00:45:20 FTP_Server: daemon is stopped
Jul  1 00:45:20 Samba_Server: smb daemon is stopped
Jul  1 00:45:22 Timemachine: daemon is stopped
Jul  1 00:45:22 avahi-daemon[2966]: WARNING: No NSS support for mDNS detected, consider installing nss-mdns!
Jul  1 00:45:22 WEBDAV_Server: daemon is stopped
Jul  1 00:45:22 STOP: About to stop: 0 0 0
Jul  1 00:45:22 STOP: bd
```

RMerl, 2021-07-01:
Asus/BCM has the HND kernels compiled with CONFIG_DEPRECATED_SYSFS and CONFIG_DEPRECATED_SYSFS_V2 enabled, which was what was requiring John's original patch to handle these using the old format.  Now for some unknown reason, routers based on the 962118 platform (which means RT-AX86U and RT-AX68U) do not have these two sets, which will now fail with the original patch applied.

Two possible solutions:

1) Add another exception for 962118 models to use the newer format
2) Remove the deprecated build options from all HND models, and revert John's patch so everything works as it was initially intended.

My fear with option 2 is that it might break other things.  I'm gonna check with upstream if they have more info on why these two deprecated options are set, and if they would be safe to revert.

ColinTaylorUK, 2021-07-01:
That's great news that you've identified the cause of the problem! I'll await the news from upstream.

In the meantime I've got a workaround by adding `/sbin/ejusb -1 1` to the end of `services-stop`.

If you want me to test any experimental builds I'd be happy to do so.

commit a754c6de (2021-07-02): shared: handle both old and new sysfs format in exec_for_host() (fixes #761); fix potential crash

Allows both old and new sysfs format to work, fixing it for all
platforms including the RT-AX86U.

Also null-terminate the readlink() result, otherwise subsequent accesses
to linkbuf may lead to unexpected results or even a router crash.
closed 2021-07-02 by RMerl