RMerl/asuswrt-merlin.ng #762: RT-AX86U - Preamble Type missing in 5GHz Wireless Professional

ColinTaylorUK, 2021-07-01:
As per the title, the option "Preamble Type" is not present in the 5GHz Wireless Professional settings page. It is present in the 2.4GHz settings.

I don't know whether this is intentional or not. I just noticed that in stock Asus firmware the option is present.

By default the nvram variable wl1_plcphdr is set to "long".

**Model**
RT-AX86U

**Firmware version**
386.2_6

fara99, 2021-07-01:
Are you sure about that? I'm using stock firmware and the option is not present  there for me. [https://i.imgur.com/l4ckle6.jpg](url)

ColinTaylorUK, 2021-07-01:
Maybe I was mistaken. I remember changing it on the stock firmware that came out of the box. Maybe I only did it for 2.4GHz and forgot about 5GHz. OTOH maybe it's something Asus changed in one of the recent firmware updates.

Short of reinstalling an old Asus firmware I don't know for sure. If I'm wrong then I apologise for wasting everyone's time.

RMerl, 2021-07-01:
That option is not available for the 5 GHz radio on Broadcom devices, this explicitly get hidden when unit is set to 1 or 2, and the device is Broadcom-based.

closed 2021-07-01 by RMerl