RMerl/asuswrt-merlin.ng #763: The web UI is unavailable

we1559, 2021-07-03:
**Describe the bug**

the web UI can't load after my router upgraded to 386.2_6
the login page will only shows the first time after the router restart.
And it will never show again before the next restart.

And the first time I login, I can only see the page but chrome shows this page is unresponse.
Some icon are seem not load properly.

**Model**

ac68u

**Firmware version**

386.2_6

**Client information**

I can reproduce it in chrome, edge, IE in my computer which is connected by the lan.
And I can also reproduce it in my Samsung galaxy S10.

**To Reproduce**

open the web config page, and it will not show anything.

**Expected behavior**

**Actual results**

**Will this error be reproduced on the official firmware?**

**Screenshots**

[image]

**Additional context**

RMerl, 2021-07-03:
Upgraded from what version?

we1559, 2021-07-03:
> Upgraded from what version?

I can't rembember, maybe 384.12？
Is there anyway I can find the original version?

RMerl, 2021-07-03:
Have you read the 386.1 release notes?  Moving from 384.xx to 386.xx requires internal database maintenance, which can take up to a few hours depending on the size of the database and the router's CPU speed.

RMerl, 2021-07-03:
Also if upgrading from a much older 384.xx release, it's probably better to do a factory default reset (including erasing the JFFS partition content), and reconfiguring from scratch, as I cannot guarantee that such old settings will be optimal with recent releases.

we1559, 2021-07-03:
Alright, I just got the information after I read the 386.1 release notes.
It might be better to show this info prominently as most people wouldn't  read the release notes of all history release.

I will wait for it response and try to do a factory default reset.

Thanks for your help.

closed 2021-07-03 by we1559