RMerl/asuswrt-merlin.ng #766: 'openvpn-event' custom script not persistent

macek-jm, 2021-07-07:
Custom script: 'openvpn-event' file disappear after reboot, router. [no problem witch other custom scripts]

**Model**

RT-AC86U

**Firmware version**

386.2_6

**Client information**

**To Reproduce**

put file to /jffs/scripts and reboot

**Expected behavior**

should stay 
**Actual results**

file dissapear ;)

**Will this error be reproduced on the official firmware?**

**Screenshots**

**Additional context**

RMerl, 2021-07-08:
Unable to reproduce here.

Try the following:

1) Post the content of /jffs/asd.log (if you have that file
2) Remove the content of /jffs/asd/ then reboot

macek-jm, 2021-07-09:
Thanks, now everything is ok[after remove content of /jffs/asd/ ],
what mistake I made?

content of my old  asd.log 

1525496710[register_feature] blockfile registered 
1525496710[register_feature] chknvram registered 
1525496710[register_feature] misc registered 
1525496710[init] Load existing signature files.
1525496710[update_signature_in_feature] Update sig in feature(blockfile)
1525496710[update_signature_in_feature] Update sig in feature(chknvram)
1525496715[chknvram_action]Unset invalid nvram,<jffs2_scripts, 0>
1624989729[check_version_from_server] Check server at Tue Jun 29 20:02:09 2021
1624989731[update_signature_in_feature] Update sig in feature(chknvram)
1624989731[check_version_from_server]update version file
1525496711[register_feature] blockfile registered 
1525496711[register_feature] chknvram registered 
1525496711[register_feature] misc registered 
1525496711[init] Load existing signature files.
1525496711[update_signature_in_feature] Update sig in feature(blockfile)
1525496711[update_signature_in_feature] Update sig in feature(chknvram)
1525496710[register_feature] blockfile registered 
1525496710[register_feature] chknvram registered 
1525496710[register_feature] misc registered 
1525496710[init] Load existing signature files.
1525496710[update_signature_in_feature] Update sig in feature(blockfile)
1525496710[update_signature_in_feature] Update sig in feature(chknvram)
1625208260[check_version_from_server] Check server at Fri Jul  2 08:44:20 2021
1625208263[_download_and_verify_file] Download file (version) fail!
1625208290[check_version_from_server] Check server at Fri Jul  2 08:44:50 2021
1625208293[_download_and_verify_file] Download file (version) fail!
1625208320[check_version_from_server] Check server at Fri Jul  2 08:45:20 2021
1625208323[_download_and_verify_file] Download file (version) fail!
1525496712[register_feature] blockfile registered 
1525496712[register_feature] chknvram registered 
1525496712[register_feature] misc registered 
1525496712[init] Load existing signature files.
1525496712[update_signature_in_feature] Update sig in feature(blockfile)
1525496712[update_signature_in_feature] Update sig in feature(chknvram)
1525496710[register_feature] blockfile registered 
1525496710[register_feature] chknvram registered 
1525496710[register_feature] misc registered 
1525496710[init] Load existing signature files.
1525496710[update_signature_in_feature] Update sig in feature(blockfile)
1525496710[update_signature_in_feature] Update sig in feature(chknvram)
1525496710[register_feature] blockfile registered 
1525496710[register_feature] chknvram registered 
1525496710[register_feature] misc registered 
1525496710[init] Load existing signature files.
1525496710[update_signature_in_feature] Update sig in feature(blockfile)
1525496710[update_signature_in_feature] Update sig in feature(chknvram)
1525496710[register_feature] blockfile registered 
1525496710[register_feature] chknvram registered 
1525496710[register_feature] misc registered 
1525496710[init] Load existing signature files.
1525496710[update_signature_in_feature] Update sig in feature(blockfile)
1525496710[update_signature_in_feature] Update sig in feature(chknvram)
1525496710[register_feature] blockfile registered 
1525496710[register_feature] chknvram registered 
1525496710[register_feature] misc registered 
1525496710[init] Load existing signature files.
1525496710[update_signature_in_feature] Update sig in feature(blockfile)
1525496710[update_signature_in_feature] Update sig in feature(chknvram)
1525496710[register_feature] blockfile registered 
1525496710[register_feature] chknvram registered 
1525496710[register_feature] misc registered 
1525496710[init] Load existing signature files.
1525496710[update_signature_in_feature] Update sig in feature(blockfile)
1525496710[update_signature_in_feature] Update sig in feature(chknvram)
1525496710[register_feature] blockfile registered 
[... 282 lines cut ...]
1525496712[register_feature] blockfile registered 
1525496712[register_feature] chknvram registered 
1525496712[register_feature] misc registered 
1525496712[init] Load existing signature files.
1525496712[update_signature_in_feature] Update sig in feature(blockfile)
1525496712[update_signature_in_feature] Update sig in feature(chknvram)
1525496713[remove_file_in_dir]Delete harmful file,/jffs/scripts/openvpn-event
1525496713[blockfile] /jffs/scripts/openvpn-event is binary.
1625768741[check_version_from_server] Check server at Thu Jul  8 20:25:41 2021
1525496712[register_feature] blockfile registered 
1525496712[register_feature] chknvram registered 
1525496712[register_feature] misc registered 
1525496712[init] Load existing signature files.
1525496712[update_signature_in_feature] Update sig in feature(blockfile)
1525496712[update_signature_in_feature] Update sig in feature(chknvram)
1525496713[remove_file_in_dir]Delete harmful file,/jffs/scripts/openvpn-event
1525496713[blockfile] /jffs/scripts/openvpn-event is binary.
1525496712[register_feature] blockfile registered 
1525496712[register_feature] chknvram registered 
1525496712[register_feature] misc registered 
1525496713[init] Load existing signature files.
1525496713[update_signature_in_feature] Update sig in feature(blockfile)
1525496713[update_signature_in_feature] Update sig in feature(chknvram)
1525496714[remove_file_in_dir]Delete harmful file,/jffs/scripts/openvpn-event
1525496714[blockfile] /jffs/scripts/openvpn-event is binary.

RMerl, 2021-07-09:
Your router was still using ASD security signatures from the stock firmware, which deletes any existing openvpn-event script.  Removing the signatures fixed it, your router should normally download Asuswrt-Merlin specific signatures after that.  No idea why it didn't, maybe something on your configuration is preventing your router from properly updating its signature files with Asuswrt-Merlin versions.

closed 2021-07-09 by RMerl