RMerl/asuswrt-merlin.ng #767: Formatting error in output of iptables fullcone option

ColinTaylorUK, 2021-07-07:
The output of `iptables-save` and `iptables` has incorrect spacing for the "fullcone" option. The output of `iptables-save` in particular creates a syntactically invalid rule should there be an attempt to reload it.

**Model**
RT-AX86U

**Firmware version**
386.2_6

**To Reproduce**
`iptables -t nat -A POSTROUTING -s 99.99.99.99 -o eth0 -j MASQUERADE --mode fullcone`

**Expected behavior**
The output of `iptables-save -t nat` and `iptables -t nat -L -v -n` should contain:

`-A POSTROUTING -s 99.99.99.99/32 -o eth0 -j MASQUERADE --mode fullcone`
and
`    0     0 MASQUERADE  all  --  *      eth0    99.99.99.99          0.0.0.0/0            mode: fullcone`

[right.txt](https://github.com/RMerl/asuswrt-merlin.ng/files/6779804/right.txt)

**Actual results**
`-A POSTROUTING -s 99.99.99.99/32 -o eth0 -j MASQUERADE--mode fullcone`
and
`    0     0 MASQUERADE  all  --  *      eth0    99.99.99.99          0.0.0.0/0           mode: fullcone`

[wrong.txt](https://github.com/RMerl/asuswrt-merlin.ng/files/6779805/wrong.txt)

**Additional context**

The offending lines are here: https://github.com/RMerl/asuswrt-merlin.ng/blob/28363cc18132f149820c6bcd543a2e384ec5d9d8/release/src/router/iptables-1.4.x/extensions/libipt_MASQUERADE.c#L169

`		printf("--mode fullcone ");`
should be
`		printf(" --mode fullcone");`

and here: https://github.com/RMerl/asuswrt-merlin.ng/blob/28363cc18132f149820c6bcd543a2e384ec5d9d8/release/src/router/iptables-1.4.x/extensions/libipt_MASQUERADE.c#L149

`		printf("mode: fullcone ");`
should be
`		printf(" mode: fullcone");`

RMerl, 2021-07-08:
Thanks, fixed with dd026dedfb .

closed 2021-07-08 by RMerl

ColinTaylorUK, 2021-07-08:
Thanks Eric.

Being pedantic the trailing space in each of those strings should also be removed to make it consistent with the rest of the output. Otherwise there's a spurious space at the end of each of those lines with isn't there for the others. But as those are the last strings output on each line it doesn't really matter and I can understand you wanting the "play it safe".