RMerl/asuswrt-merlin.ng #769: RT-AX58U 2.4 GHz radio very low TX Rate

MeFoX999, 2021-07-08:
2.4 GHz radio TX rate drops to 5 Mbps, and real transfer is about 2 Mbps. It's happen about 1 minute after reboot and stays on this level. Any transmission causing a drop. I had this problem on stock firmware from January. I've decided to check Merlin, but it didn't resolve this issue. Now, on recent stock firmware everything is OK, on Merlin not. No matter how far 2.4 GHz device is, any settings not resolving issue. Even after full reset and basic configuration problem showing up immediately.

dave14305, 2021-07-08:
Is the stock version a higher number than 42095? Merlin is on 42095.

Wireless code is closed-source and outside of Merlin’s control.

RMerl, 2021-07-08:
Wireless is closed source and outside of my control.  You are getting different results because you are comparing different driver versions.  You will have to wait until new GPL code is available from Asus to bring the driver up to the same version.

closed 2021-07-08 by RMerl