RMerl/asuswrt-merlin.ng #771: dhcp-event should identify if it was called from udhcpc or dhcp6c

jtrahan, 2021-07-11:
currently the event sends all events for both udhcpc and dhcp6c. but there is no direct way to tell which one originated the event. I currently have to use on of the IPV6 environment variable to determine which dhcp daemon called the script. It would be helpful to either:
A. have a new argument passed to the script to identify which process made the call
B. have an environment variable identifying which one was called
C. split out the events so that there is a dhcpc-event and a dhcp6c-event

JackMerlin, 2021-07-12:
Please do not use Github's Issue tracker for support requests.  Use the support forums at [SNBForums](https://www.snbforums.com/forums/asuswrt-merlin.42/) instead.  
The issue tracker is for development/bug tracking purposes only!

https://raw.githubusercontent.com/RMerl/asuswrt-merlin.ng/master/SUPPORT.md

jtrahan, 2021-07-12:
My Apologizes on that. I'll go that route next time. However I do see this as a defect that should be addressed since it's triggering dhcp events for both ipv4 and ipv6 events and a misleading trigger could potentially cause unexcepted behaviors depending on how someone writes a handler for the event.  This is what happened to me as I discovered that I had my dhcp events wired for my original ipv4 setup and not realizing the same event was used for ipv6 it started causing those unexpected behaviors when I started experimenting and utilizing ipv6.