RMerl/asuswrt-merlin.ng #772: after each reboot, only first openvpn client can connect,all other client authentication failed.

haohetao, 2021-07-12:
**Describe the bug**
after each reboot, only first openvpn client can connect,all other client authentication failed.

**Model**
ax88u

**Firmware version**
386.2_4

**Client information**

**To Reproduce**

**Expected behavior**

**Actual results**

**Will this error be reproduced on the official firmware?**

**Screenshots**

**Additional context**

JackMerlin, 2021-07-12:
#754 the same issue, please add your details to the original issue #754 instead of opening a new issue.

haohetao, 2021-07-13:
This problem has existed for several months. Why hasn't it been solved.
@JackMerlin

haohetao, 2021-07-13:
Deleting and re adding VPN username and password can solve the problem temporarily.

JackMerlin, 2021-07-16:
Please note that I am not a developer, I can't answer your technical issues. I'm very sorry.