RMerl/asuswrt-merlin.ng #773: Port forwarding not working properly with NAT acceleration enabled

beefjerky9, 2021-07-15:
**Describe the bug**

When NAT acceleration is enabled, some of the port forwarding will not work. It seems to be random which ports are affected for each user. In my case, it is ports 80 and 3389. Disabling NAT acceleration will successfully work around the issue. This seems to have begun in firmware version 386.1_2.

**Model**

RT-AC3100

**Firmware version**

386.2_6

**Client information**

This is not linked to any specific client.

**To Reproduce**

Create some port forwards and leave NAT acceleration enabled.

**Expected behavior**

All of the port forwards working.

**Actual results**

Some of the port forwards do not work.

**Will this error be reproduced on the official firmware?**

n/a

**Screenshots**

n/a

**Additional context**

n/a

RMerl, 2021-07-15:
So far I'm unable to reproduce, it's working in my test with a web server sitting behind an RT-AC88U (which uses the same SDK/hardware acceleration driver as the RT-AC3100).

Can you find a constant that can more accurately trigger it?  I know I reverted some code a few months ago related to that as I was told it was no longer necessary.

beefjerky9, 2021-07-15:
I'm not sure what triggers it. Perhaps it's a certain combination of ports? Here's a screenshot of my port configuration; maybe you can have some luck trying to emulate it.

[image]

That said, I'm definitely not the only one. Here are two threads from snbforums with multiple posters with the same issue:
https://www.snbforums.com/threads/port-forwarding-only-working-on-some-ports-ax86u.72662/
https://www.snbforums.com/threads/port-forwarding-broken-with-hw-nat-acceleration.72146/

RMerl, 2021-07-15:
Debugging anything related to hardware acceleration is next to impossible because it's a blackbox that even Asus don't have access to the internals (only Broadcom does).

I could try reverting the change I did back then and hope that it fixes it, but it will be hard to confirm without a sure way to reproduce the issue.

beefjerky9, 2021-07-15:
If you wanted me to try a version of the firmware with that change reversed, I'd be happy to try it and let you know the results. This way it can be tested on a router that shows the issue.

RMerl, 2021-07-15:
Actually, it seems I never went ahead with the planned change, so that's not it.  That would only leave a potential bug within the CTF module, which is completely outside of my control.

beefjerky9, 2021-07-15:
Well, that's disappointing, but not your fault. Is there any way for us to report it to the appropriate parties?

RMerl, 2021-07-15:
You will need to reproduce it with the latest stock firmware (to ensure the same issue exists with stock firmware and that it hasn't been fixed yet in their latest release), and then use the Feedback form on the webui to report the issue.

beefjerky9, 2021-07-15:
Okay, I confirmed that it still existed in the current stock firmware and submitted it to them using the feedback form. I will post in the threads on snbforums encouraging others to do the same. If they get multiple people complaining about the issue, I'd say they're more likely to care. Thanks for all the feedback!

Urbanshenkie, 2021-07-22:
Port forwarding with Nat acceleration is not recommended. Read it here.

https://routerguide.net/nat-acceleration-on-or-off/

beefjerky9, 2021-07-22:
The page you linked to really doesn't say what you claim it does. It lists a few possible limitations, but it by no means recommends to not use it. In fact, the page recommends leaving it set to auto. It also says that it may be needed when a person has fast enough internet. I have gigabit fiber, and I do seem to notice a slowdown of peak speeds compared to with it on.

The port forwarding worked perfectly well with earlier firmware and NAT acceleration enabled. I hope Asus fixes it. I've thought about rolling back to earlier firmware where it was working properly, but then I open myself up to possible security issues.