RMerl/asuswrt-merlin.ng #774: Traffic Monitor tab and info in System Status from LAN IP downwards stop working

cthart, 2021-07-16:
Quite regularly for me, after being logged in for a while, having closed the browser tab/window and then returning to the web interface -- which doesn't require me logging in again -- the Traffic Monitor tab of the Traffic Analyzer stops working:
https://imgur.com/hH29TpS.png

Note that the Scale dropdown is empty, and there's no graph.

When that happens, the LAN IP and all fields below it in System Status are all empty:
https://imgur.com/rYERzAT.png

Logging out and logging in again directly fixes the problem immediately.

JackMerlin, 2021-07-16:
Please try to follow the issue template to provide more useful information.

Since it looks a browser problem, let us know what browser version you are using, and try to install a new browser without the browser extension to test it.

Finally, don't forget any of the error outputs in the browser console.

cthart, 2021-07-16:
Sorry, for some reason I didn’t get the temple when I created the issue. Will edit.

cthart, 2021-07-17:
Edited.

**Summary of bug**: update.cgi and appGet.cgi think that the session is no longer valid while the rest of the interface thinks the session is still OK.