RMerl/asuswrt-merlin.ng #775: Mixed Content error

cthart, 2021-07-16:
**Describe the bug**
jquery.js:5 Mixed Content: The page at 'https://<redacted>.asuscomm.com:8443/' was loaded over HTTPS, but requested an insecure XMLHttpRequest endpoint 'http://nw-dlcdnet.asus.com/plugin/js/extend_custom_icon.json'. This request has been blocked; the content must be served over HTTPS.

**Model**
RT-AC68U

**Firmware version**
386.2_6

**Client information**
Google Chrome 91.0.4472.164

**To Reproduce**
Load the web interface in the browser and view the Network tab in Developer Tools.

**Expected behavior**
Use HTTPS instead.

**Actual results**
As above.

**Will this error be reproduced on the official firmware?**
Haven't used original firmware.

**Screenshots**
n/a

**Additional context**
n/a

JackMerlin, 2021-07-16:
This is from the upstream problem, please try to reproduce the problem under the official firmware and feedback to ASUS

RMerl, 2021-07-16:
Some of the webui queries sent to Asus are done over http because until very recently that server didn't have a valid SSL certificate (it used an AWS SAN, causing validation errors).  They are currently in the process of fixing this, so it will be resolved on my end once their server is ready for it, and they update all links within the code to use https.

closed 2021-07-16 by RMerl