RMerl/asuswrt-merlin.ng #778: DDNS update from GUI deletes all the other added configurations in the `/etc/inadyn.conf` file

maicol07, 2021-07-22:
**Describe the bug**
Updating the DDNS from the GUI deletes all the added configurations in the `/etc/inadyn.conf` file

**Model**
DSL-AX82U (@gnuton's fork)

**Firmware version**
386.03_1-gnuton0_alpha3

**Client information**
Browser: Firefox 91 x64

**To Reproduce**
1. Add custom config (other providers) to the `/etc/inadyn.conf` file
2. Update DDNS from the GUI
3. Check the file from step 1

**Expected behavior**
The configuration of the DDNS from the GUI gets updated and doesn't overwrite my configs

**Actual results**
Custom/added configurations are overwritten by the updated DDNS from the GUI

**Will this error be reproduced on the official firmware?**

RMerl, 2021-07-22:
Working as intended.  Anything in /etc is dynamically generated by the firmware, and does not survive reboots.  You should not modify anything there, and instead rely on the existing postconf system.

closed 2021-07-22 by RMerl

maicol07, 2021-07-22:
> Working as intended.  Anything in /etc is dynamically generated by the firmware, and does not survive reboots.  You should not modify anything there, and instead rely on the existing postconf system.

Is there a guide on that system?

RMerl, 2021-07-22:
This project's Wiki, link is at the top.