RMerl/asuswrt-merlin.ng #781: QR Code not scannable

maicol07, 2021-07-23:
**Describe the bug**
QR Code cannot be scanned by readers (i.e. Google Lens) at the moment since they don't have enough white border and it can't be recognized. Works if you save the QR code image and open it with a photo viewer on your PC

**Model**
DSL-AX82U

**Firmware version**
386.03_1-gnuton0_alpha3

**Client information**
Browser: Firefox 91 x64

**To Reproduce**
1. Go to Network map
2. In sidebar click Wireless and then Show QR Code
3. Try to scan it with a reader (i.e. Google Lens)

Workaround:
1. Right click --> Save image on QR Code
2. Open the image with image viewer (i.e. Photos on  Windows)
3. Try to scna it again

**Expected behavior**
QR Code can be scanned without issues

**Actual results**
QR Code can't be scanned

**Will this error be reproduced on the official firmware?**

RMerl, 2021-07-23:
Works for me with two different devices, and also none of the testers reported anything similar over the past 2+ months of betatesting.  Most likely an issue with your client, or your display.

closed 2021-07-23 by RMerl

RMerl, 2021-07-23:
I'll increase the size of the quiet zone, this may help with scanners that have trouble detecting the code.

RMerl, 2021-07-23:
Adding a lighter quiet zone does make scanning faster on my phone (I no longer need to be as close to the image), it should also help with your scanning tool if it was having trouble picking it up at all.

maicol07, 2021-07-23:
I've also tried with another browser and another device and I got the same issue...

RMerl, 2021-07-23:
Could be due to the camera quality, or the contrast on your display (I do have a colour-calibrated display, so there might be a larger contrast difference between the code and the quiet zone background).

Should work better for you now that I made the quiet zone a lighter colour.  With my phone I was now able to scan the code from even 30-40 cm away from my monitor, instead of having to be within 10-15 cm.

maicol07, 2021-07-24:
It's better indeed. Now the QR gets scanned instantly! Thank you!